luigifcruz/CyberEther #11: Add `INSTALL.md`

Teque5, 2022-09-26:
The `README` doesn't really have instructions.

I *think* it's:
```bash
mkdir build;cd build
meson build ../
meson install
<Ubuntu 22.04 compile error>
????
```
My output:
<pre>meson install
ninja: Entering directory `/home/kal29868/Source/cyberether/build&apos;
[2/34] Compiling C++ object subproject...ibjetstream.so.p/src_backend_base.cc.o
<font color="#C01C28">FAILED: </font>subprojects/jetstream/libjetstream.so.p/src_backend_base.cc.o 
c++ -Isubprojects/jetstream/libjetstream.so.p -Isubprojects/jetstream -I../subprojects/jetstream -Isubprojects/jetstream/include -I../subprojects/jetstream/include -I../subprojects/jetstream/include/jetstream/backend/devices/metal -I../subprojects/jetstream/include/jetstream/render/tools -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -std=c++20 -g -fPIC -pthread -MD -MQ subprojects/jetstream/libjetstream.so.p/src_backend_base.cc.o -MF subprojects/jetstream/libjetstream.so.p/src_backend_base.cc.o.d -o subprojects/jetstream/libjetstream.so.p/src_backend_base.cc.o -c ../subprojects/jetstream/src/backend/base.cc
In file included from <b>../subprojects/jetstream/src/backend/base.cc:1</b>:
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:31:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   31 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:32:18:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>initialize&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   32 |     const Result <font color="#C01C28"><b>initialize&lt;Device::CPU&gt;</b></font>(const Config&amp; config) {
      |                  <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:39:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   39 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:40:18:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>destroy&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   40 |     const Result <font color="#C01C28"><b>destroy&lt;Device::CPU&gt;</b></font>() {
      |                  <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:47:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   47 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:48:17:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>state&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   48 |     const auto&amp; <font color="#C01C28"><b>state&lt;Device::CPU&gt;</b></font>() {
      |                 <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:85:29:</b> <font color="#A347BA"><b>warning: </b></font>‘<b>visibility</b>’ attribute ignored on non-class types [<font color="#A347BA"><b>-Wattributes</b></font>]
   85 | Instance&amp; JETSTREAM_API Get(<font color="#A347BA"><b>)</b></font>;
      |                             <font color="#A347BA"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:88:33:</b> <font color="#A347BA"><b>warning: </b></font>‘<b>visibility</b>’ attribute ignored on non-class types [<font color="#A347BA"><b>-Wattributes</b></font>]
   88 | const auto&amp; JETSTREAM_API State(<font color="#A347BA"><b>)</b></font> {
      |                                 <font color="#A347BA"><b>^</b></font>
[7/34] Compiling C++ object subproject...am/libjetstream.so.p/src_instance.cc.o
<font color="#C01C28">FAILED: </font>subprojects/jetstream/libjetstream.so.p/src_instance.cc.o 
c++ -Isubprojects/jetstream/libjetstream.so.p -Isubprojects/jetstream -I../subprojects/jetstream -Isubprojects/jetstream/include -I../subprojects/jetstream/include -I../subprojects/jetstream/include/jetstream/backend/devices/metal -I../subprojects/jetstream/include/jetstream/render/tools -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -std=c++20 -g -fPIC -pthread -MD -MQ subprojects/jetstream/libjetstream.so.p/src_instance.cc.o -MF subprojects/jetstream/libjetstream.so.p/src_instance.cc.o.d -o subprojects/jetstream/libjetstream.so.p/src_instance.cc.o -c ../subprojects/jetstream/src/instance.cc
In file included from <b>../subprojects/jetstream/include/jetstream/instance.hh:8</b>,
                 from <b>../subprojects/jetstream/src/instance.cc:1</b>:
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:31:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   31 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:32:18:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>initialize&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   32 |     const Result <font color="#C01C28"><b>initialize&lt;Device::CPU&gt;</b></font>(const Config&amp; config) {
      |                  <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:39:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   39 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:40:18:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>destroy&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   40 |     const Result <font color="#C01C28"><b>destroy&lt;Device::CPU&gt;</b></font>() {
      |                  <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:47:14:</b> <font color="#C01C28"><b>error: </b></font>explicit specialization in non-namespace scope ‘<b>class Jetstream::Backend::Instance</b>’
   47 |     template&lt;<font color="#C01C28"><b>&gt;</b></font>
      |              <font color="#C01C28"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:48:17:</b> <font color="#C01C28"><b>error: </b></font>template-id ‘<b>state&lt;Jetstream::Device::CPU&gt;</b>’ in declaration of primary template
   48 |     const auto&amp; <font color="#C01C28"><b>state&lt;Device::CPU&gt;</b></font>() {
      |                 <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:85:29:</b> <font color="#A347BA"><b>warning: </b></font>‘<b>visibility</b>’ attribute ignored on non-class types [<font color="#A347BA"><b>-Wattributes</b></font>]
   85 | Instance&amp; JETSTREAM_API Get(<font color="#A347BA"><b>)</b></font>;
      |                             <font color="#A347BA"><b>^</b></font>
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:88:33:</b> <font color="#A347BA"><b>warning: </b></font>‘<b>visibility</b>’ attribute ignored on non-class types [<font color="#A347BA"><b>-Wattributes</b></font>]
   88 | const auto&amp; JETSTREAM_API State(<font color="#A347BA"><b>)</b></font> {
      |                                 <font color="#A347BA"><b>^</b></font>
[... 25 lines cut ...]
<b>../subprojects/jetstream/include/jetstream/backend/base.hh:85:29:</b> <font color="#A347BA"><b>warning: </b></font>‘<b>visibility</b>’ attribute ignored on non-class types [<font color="#A347BA"><b>-Wattributes</b></font>]
   85 | Instance&amp; JETSTREAM_API Get(<font color="#A347BA"><b>)</b></font>;
      |                             <font color="#A347BA"><b>^</b></font>
[... cut ...]
<b>/usr/include/c++/11/bits/shared_ptr_base.h:1299:16:</b> <font color="#2AA1B3"><b>note: </b></font>‘<b>std::__shared_ptr&lt;_Tp, _Lp&gt;::operator bool() const [with _Tp = Jetstream::Render::Texture; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]</b>’ declared here
 1299 |       explicit <font color="#2AA1B3"><b>operator</b></font> bool() const noexcept
      |                <font color="#2AA1B3"><b>^~~~~~~~</b></font>
[11/34] Compiling C++ object subprojec...n-310-x86_64-linux-gnu.so.p/index.cc.o
ninja: build stopped: subcommand failed.
</pre>

Flamewires, 2022-09-26:
similar issue on fully updated arch using gcc version 12.2.0 (GCC). I do not have any GPU libs installed on this host so no CUDA APIs.

```
$ meson clean .
The Meson build system
Version: 0.63.2
Source dir: /home/USER/CyberEther
Build dir: /home/USER/CyberEther/clean
Build type: native build
Project name: cyberether
Project version: 0.0.0
C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (GCC) 12.2.0")
C++ linker for the host machine: c++ ld.bfd 2.39.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Run-time dependency threads found: YES

Executing subproject samurai 

samurai| Project name: samurai
samurai| Project version: 1.0.0
samurai| C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (GCC) 12.2.0")
samurai| C++ linker for the host machine: c++ ld.bfd 2.39.0
samurai| Message: include
samurai| Message: /usr/local
samurai| Dependency threads found: YES unknown (cached)
samurai| Found pkg-config: /usr/bin/pkg-config (1.8.0)
samurai| Run-time dependency limesuite found: YES 20.10.0-ga45e482d
samurai| Run-time dependency libairspy found: YES 1.0
samurai| Run-time dependency libairspyhf found: YES 1.6
samurai| Configuring config.hpp using configuration
menssamurai| Program python3 found: YES (/usr/bin/python3)
samurai| Found CMake: /usr/bin/cmake (3.24.2)
samurai| Run-time dependency pybind11 found: YES 2.10.0
samurai| Program lib.py found: YES (/home/USER/CyberEther/subprojects/samurai/test/lib.py)
samurai| Build targets in project: 4
samurai| Subproject samurai finished.

Executing subproject jetstream 

jetstream| Project name: jetstream
jetstream| Project version: 0.1.0
jetstream| C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (GCC) 12.2.0")
jetstream| C++ linker for the host machine: c++ ld.bfd 2.39.0
jetstream| Dependency threads found: YES unknown (cached)
jetstream| Run-time dependency fmt found: YES 9.1.0
jetstream| Run-time dependency CUDA (modules: cudart, dl) found: NO (tried system)
jetstream| Run-time dependency fftw3f found: YES 3.3.10
jetstream| Run-time dependency fftw3 found: YES 3.3.10
jetstream| Run-time dependency appleframeworks found: NO (tried framework)
jetstream| Run-time dependency glfw3 found: YES 3.3.8
jetstream| Configuring jetstream_config.hh using configuration
jetstream| Build targets in project: 5
jetstream| Subproject jetstream finished.

Build targets in project: 6

jetstream 0.1.0

  Loaders
    CUDA      : NO
    FFTW      : YES
    Metal     : NO
    GLFW      : YES

  Backend (Compute)
    CPU       : YES
    CUDA      : NO
    FFTW      : YES
    Metal     : NO
[... 157 lines cut ...]
../subprojects/jetstream/include/jetstream/render/base/surface.hh: In member function ‘constexpr const Jetstream::Render::Size2D<long unsigned int>& Jetstream::Render::Surface::size() const’:
../subprojects/jetstream/include/jetstream/render/base/surface.hh:29:20: error: call to non-‘constexpr’ function ‘std::__shared_ptr<_Tp, _Lp>::operator bool() const [with _Tp = Jetstream::Render::Texture; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’
   29 |         if (config.framebuffer) {
      |             ~~~~~~~^~~~~~~~~~~
In file included from /usr/include/c++/12.2.0/bits/shared_ptr.h:53,
                 from /usr/include/c++/12.2.0/memory:77,
                 from /usr/include/fmt/format.h:40,
                 from ../subprojects/jetstream/include/jetstream/logger.hh:6,
                 from ../subprojects/jetstream/include/jetstream/render/base.hh:4:
/usr/include/c++/12.2.0/bits/shared_ptr_base.h:1669:16: note: ‘std::__shared_ptr<_Tp, _Lp>::operator bool() const [with _Tp = Jetstream::Render::Texture; __gnu_cxx::_Lock_policy _Lp = __gnu_cxx::_S_atomic]’ declared here
 1669 |       explicit operator bool() const noexcept
      |                ^~~~~~~~
[24/40] Compiling C++ object subprojects/jetstream/libjetstream.so.p/src_render_tools_imgui_widgets.cpp.o
../subprojects/jetstream/src/render/tools/imgui_widgets.cpp: In function ‘bool ImGui::TempInputScalar(const ImRect&, ImGuiID, const char*, ImGuiDataType, void*, const char*, const void*, const void*)’:
../subprojects/jetstream/src/render/tools/imgui_widgets.cpp:3406:67: warning: bitwise operation between different enumeration types ‘ImGuiInputTextFlags_’ and ‘ImGuiInputTextFlagsPrivate_’ is deprecated [-Wdeprecated-enum-enum-conversion]
 3406 |     ImGuiInputTextFlags flags = ImGuiInputTextFlags_AutoSelectAll | ImGuiInputTextFlags_NoMarkEdited;
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../subprojects/jetstream/src/render/tools/imgui_widgets.cpp: In function ‘bool ImGui::CollapsingHeader(const char*, bool*, ImGuiTreeNodeFlags)’:
../subprojects/jetstream/src/render/tools/imgui_widgets.cpp:6164:54: warning: bitwise operation between different enumeration types ‘ImGuiTreeNodeFlags_’ and ‘ImGuiTreeNodeFlagsPrivate_’ is deprecated [-Wdeprecated-enum-enum-conversion]
 6164 |         flags |= ImGuiTreeNodeFlags_AllowItemOverlap | ImGuiTreeNodeFlags_ClipLabelForTrailingButton;
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[26/40] Compiling C++ object subprojects/samurai/python/samurai.cpython-310-x86_64-linux-gnu.so.p/index.cc.o
ninja: build stopped: subcommand failed.

```

luigifcruz, 2022-09-26:
Hey, thanks for opening this issue! Your installation commands are correct, but it looks like you're trying to run this on Linux. The current version doesn't implement a Linux-compatible render (only Apple Metal for now). If you would like to try an older version that supports Linux you could try 35693631c3110dfab0327599f15922fdf0028005. That commit looks like it builds fine. Unfortunately, this will only work with Airspy or LimeSDR without GNU Radio.

```
$ git clone https://github.com/luigifcruz/CyberEther.git
$ cd CyberEther
$ git checkout 35693631c3110dfab0327599f15922fdf0028005
$ meson build -Dbuildtype=release
$ cd build
$ ninja
$ ./cyberether
```

I'm currently working on writing a Vulkan backend for the render that would support Linux targets. It should be available in the very near future.

Flamewires, 2022-09-26:
Sounds good. Happen to have airspy so gave it a go, any ideas on them missing font issue?

```

[USER@HOST build]$ airspy_info 
airspy_lib_version: 1.0.10

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc10-3-g7120e77 2018-04-28
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x62CC68FF3575385B
Supported sample rates:
	10.000000 MSPS
	2.500000 MSPS
Close board 1
[USER@HOST build]$ cyberether 
Welcome to CyberEther!
cyberether: ../subprojects/render/src/tools/imgui_draw.cpp:2159: ImFont* ImFontAtlas::AddFontFromFileTTF(const char*, float, const ImFontConfig*, const ImWchar*): Assertion `(0) && "Could not load font file!"' failed.
Aborted (core dumped)
```

luigifcruz, 2022-09-26:
@Flamewires Yup, I forgot about that. Just run the command below inside the directory you are running `./cyberether`. This will download the necessary font file. The current `main` has this built-in, so no download is necessary.

```
$ wget https://github.com/luigifcruz/CyberEther/raw/02513b8e3bbe5c0ffdf5ed4b9eaf15f958cb140a/subprojects/jetstream/resources/B612Mono-Regular.ttf
```

Flamewires, 2022-09-26:
Seemed to work, thanks. Will keep an eye out for the vulkan backend of main branch.

texadactyl, 2022-09-26:
Note:
```
samurai| Run-time dependency pybind11 found: NO (tried pkgconfig and cmake)

subprojects/samurai/python/meson.build:6:3: ERROR: Dependency "pybind11" not found, tried pkgconfig and cmake
```
I have pybind11 installed as a Python package but not centrally where samurai expected to find it (/usr/local*).

Fix:
```
pip uninstall -y pybind11
pip install "pybind11[global]"
```
Samurai - check!  On to the next hiccup.  (;

texadactyl, 2022-09-26:
For jetstream, meson.build dependency "threads" found but not "benchmark".
```
Project name: jetstream
Project version: 0.0.0
C++ compiler for the host machine: c++ (gcc 11.2.0 "c++ (Ubuntu 11.2.0-19ubuntu1) 11.2.0")
C++ linker for the host machine: c++ ld.bfd 2.38
C++ compiler for the build machine: c++ (gcc 11.2.0 "c++ (Ubuntu 11.2.0-19ubuntu1) 11.2.0")
C++ linker for the build machine: c++ ld.bfd 2.38
Dependency threads found: YES unknown (cached)
Pkg-config binary for 1 is cached.
Determining dependency 'benchmark' with pkg-config executable '/usr/bin/pkg-config'
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --modversion benchmark` -> 1

CMake binary for 1 is cached.
Preliminary CMake check failed. Aborting.
Run-time dependency benchmark found: NO (tried pkgconfig and cmake)

subprojects/jetstream/meson.build:11:0: ERROR: Dependency "benchmark" not found, tried pkgconfig and cmake
```
Fix:
Install benchmark globally from ```https://github.com/google/benchmark```

texadactyl, 2022-09-26:
```
render| Run-time dependency fmt found: NO (tried pkgconfig and cmake)

subprojects/render/meson.build:11:0: ERROR: Dependency "fmt" not found, tried pkgconfig and cmake
```
Fix:
* Download fmt from https://fmt.dev/latest/index.html.
* Extract tar.gz.
* cd fmt
* cmake .
* make
* sudo make install

Then, ```meson build -Dbuildtype=release``` proceeds to the end successfully.

texadactyl, 2022-09-26:
@luigifcruz 
Followed your instructions as stated to use 3569363 on my Xubuntu 22.04.1 LTS laptop.

Ninja display:
```
[11/36] Compiling C++ object subprojects/render/librender.so.p/src_base.cc.o
FAILED: subprojects/render/librender.so.p/src_base.cc.o 
c++ -Isubprojects/render/librender.so.p -Isubprojects/render -I../subprojects/render -I/usr/local/include -I../subprojects/render/include/render/tools -Isubprojects/render/include -I../subprojects/render/include -Isubprojects/jetstream/include -I../subprojects/jetstream/include -Isubprojects/jetstream -I../subprojects/jetstream -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -std=c++17 -O3 -fPIC -pthread -MD -MQ subprojects/render/librender.so.p/src_base.cc.o -MF subprojects/render/librender.so.p/src_base.cc.o.d -o subprojects/render/librender.so.p/src_base.cc.o -c ../subprojects/render/src/base.cc
In file included from ../subprojects/render/src/base.cc:1:
../subprojects/render/include/render/base.hpp: In function ‘auto* Render::Get()’:
../subprojects/render/include/render/base.hpp:103:1: error: no return statements in function returning ‘auto*’
  103 | }
      | ^
../subprojects/render/include/render/base.hpp:103:1: note: only plain ‘auto’ return type can be deduced to ‘void’
../subprojects/render/include/render/base.hpp: In instantiation of ‘auto Render::Create(const T&) [with T = Render::Surface::Config]’:
../subprojects/render/include/render/base.hpp:114:18:   required from here
../subprojects/render/include/render/base.hpp:110:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
  110 |     return Get()->newMember(config);
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘Render::Result Render::Create()’:
../subprojects/render/src/base.cc:63:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   63 |     return Get()->create();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘Render::Result Render::Destroy()’:
../subprojects/render/src/base.cc:67:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   67 |     return Get()->destroy();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘Render::Result Render::Begin()’:
../subprojects/render/src/base.cc:71:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   71 |     return Get()->begin();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘Render::Result Render::End()’:
../subprojects/render/src/base.cc:75:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   75 |     return Get()->end();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘Render::Result Render::Synchronize()’:
../subprojects/render/src/base.cc:79:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   79 |     return Get()->synchronize();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘bool Render::KeepRunning()’:
../subprojects/render/src/base.cc:83:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   83 |     return Get()->keepRunning();
      |            ~~~^~
../subprojects/render/src/base.cc: In function ‘bool Render::HasCudaInterop()’:
../subprojects/render/src/base.cc:87:15: error: use of ‘auto* Render::Get()’ before deduction of ‘auto’
   87 |     return Get()->hasCudaInterop();
      |            ~~~^~
[20/36] Compiling C++ object subprojects/samurai/python/samurai.cpython-310-x86_64-linux-gnu.so.p/index.cc.o
ninja: build stopped: subcommand failed.
```

Maybe this is a Linux gcc version issue?
I am using version (Ubuntu 11.2.0-19ubuntu1) 11.2.0.

texadactyl, 2022-09-27:
@Teque5 
Having an INSTALL.md sepearate from README.md makes a lot of sense.  I am reviewing my own github repos.

texadactyl, 2022-12-20:
@luigifcruz 
There seems to be incompatibilities now (since SoapySDR) and 35693631c3110dfab0327599f15922fdf0028005.
I'll give CyberEther another look when Linux is supportable w/o a checkout.  No rush.  I am busy too.

luigifcruz, 2022-12-27:
Thanks all for the discussion! I'm now working on advancing upstream to support Linux again. Let's see how it will go.

texadactyl, 2022-12-29:
Boa sorte!

nzqo, 2023-04-05:
A section about how to integrate the SDR would be nice.
I am currently trying to get CyberEther running with a USRP b200 on a mac M1, but am having problems installing the corresponding SoapySDR module (which I understand is required?).
The error is also almost a year old with apparently no responses from the maintainers https://github.com/pothosware/SoapyUHD/issues/50.

I saw that you had CyberEther running with a different UHD device, was there anything special you did?

luigifcruz, 2023-04-25:
Tracking Vulkan implementation (aka Linux support) in #21.

nzqo, 2023-10-28:
@luigifcruz 
Probably relevant here: I cant build CyberEther with the dependencies listed, and I cant tell which of the missing ones are optional and which arent. If you could update the README that would be great :)

```log
The Meson build system
Version: 1.2.2
Source dir: /Users/nzqo/Development/CyberEther
Build dir: /Users/nzqo/Development/CyberEther/build/build
Build type: native build
Project name: cyberether
Project version: 0.0.0
C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.0.40.1)")
C++ linker for the host machine: c++ ld64 1015.7
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency threads found: YES

Executing subproject jetstream 

jetstream| Project name: jetstream
jetstream| Project version: 0.1.0
jetstream| C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.0.40.1)")
jetstream| C++ linker for the host machine: c++ ld64 1015.7
jetstream| Dependency threads found: YES unknown (cached)
jetstream| Found pkg-config: /opt/homebrew/bin/pkg-config (0.29.2)
jetstream| Run-time dependency fmt found: YES 10.1.0
jetstream| Run-time dependency CUDA (modules: cudart_static, dl) found: NO (tried system)
jetstream| Run-time dependency fftw3f found: YES 3.3.10
jetstream| Run-time dependency appleframeworks found: YES (MetalKit, Metal, Foundation)
jetstream| Program glslangValidator found: NO
jetstream| Program spirv-cross found: NO
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Run-time dependency glfw3 found: YES 3.3.8
jetstream| Run-time dependency soapysdr found: YES 0.8.1
jetstream| Run-time dependency vulkan found: NO (tried pkgconfig and system)
jetstream| Program glslangValidator found: NO
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: NO
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: NO
jetstream| Program spirv-cross found: NO
jetstream| Program naga found: NO
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)

../subprojects/jetstream/src/render/meson.build:15:4: ERROR: Problem encountered: Failed to find a compatible render. Check your dependencies and try again.
```

tomverbeure, 2023-10-29:
@nzqo 
> ../subprojects/jetstream/src/render/meson.build:15:4: ERROR: Problem encountered: Failed to find a compatible render. Check your dependencies and try again.
> ```

I was able to get past this issue by doing `brew install vulkan-tools`.

But then I got stuck on this:

```
meson.build:1:0: ERROR: Unable to detect linker for compiler `c++ -Wl,--version`
stdout:
stderr: collect2 version 13.2.0
/usr/bin/ld -syslibroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/ -dynamic -arch arm64 -platform_version macos 13.0.0 0.0 -o a.out -L/opt/homebrew/Cellar/gcc/13.2.0/bin/../lib/gcc/current/gcc/aarch64-apple-darwin22/13 -L/opt/homebrew/Cellar/gcc/13.2.0/bin/../lib/gcc/current/gcc -L/opt/homebrew/Cellar/gcc/13.2.0/bin/../lib/gcc/current/gcc/aarch64-apple-darwin22/13/../../.. --version -lstdc++ -lemutls_w -lgcc -lSystem -lgcc -no_compact_unwind -rpath @loader_path -rpath /opt/homebrew/Cellar/gcc/13.2.0/lib/gcc/current/gcc/aarch64-apple-darwin22/13 -rpath /opt/homebrew/Cellar/gcc/13.2.0/lib/gcc/current/gcc -rpath /opt/homebrew/Cellar/gcc/13.2.0/lib/gcc/current
ld: unknown options: --version
collect2: error: ld returned 1 exit status
```

Tom

luigifcruz, 2023-11-14:
@tomverbeure Thanks for testing installing it! Are you on macOS Sonoma? Looks like meson is having some problems with the DYLIB path with the new Xcode. 

There is a meson issue tracking this problem, but they are very slow at implementing fixes. https://github.com/mesonbuild/meson/issues/12288

luigifcruz, 2024-01-05:
Thanks all for the input! Detailed installation instructions are now on the README.

closed 2024-01-05 by luigifcruz