luigifcruz/CyberEther #12: Browser use

karatemir, 2022-09-27:
Hi there,

watched your GrCon presentation online, great work! In your presentation you mention
> Rendered frame can be attached to any window (Qt, Cocoa, MTKView, etc)
> TL;DR: It’s easy to use on iOS, Android, macOS, or even a browser

What is the recommended way for visualizing on a browser? websockets? wasm? any examples show casing how to use?

luigifcruz, 2022-09-27:
Hey, thanks for creating this issue! The current branch can't be run on the browse. This is because I added some new features that WebGL wouldn't be able to support. To solve this, I'm planning to write a new WebGPU backend for browsers. The problem is that WebGPU isn't available for Chrome and Firefox as of now. I'm hoping they will make it available by January 2023.

Regarding the architectural question, CyberBrowser worked using WebAssembly with multi-threading enabled. The communication with the SDR was made by a [LibUSB to WebUSB wrapper](https://github.com/luigifcruz/webusb-libusb) that I wrote some time ago. That repository also has some examples, but it requires a patched version of Emscripten to work (due to some synchronization requirements imposed by WebUSB limitations).

karatemir, 2022-12-24:
@luigifcruz Hey Luigi, had a chance to revisit this repo after seeing your recent tweet about running CyberEther on ios. (great stuff btw!) In your previous reply you mention CyberBrowser. Is this the radio-blast stuff [here](https://github.com/luigifcruz/CyberRadioBlast) or is it a completely separate project?

luigifcruz, 2022-12-31:
Hey, @karatemir! It's a different project. I wrote it as a proof of concept for my WebUSB to libusb translation layer. It did work well but there are limitations. I'm planning to start the WebGPU backend for CyberEther as soon as the technology is widely available and the limitations (multi-thread) are solved.

closed 2022-12-31 by karatemir