luigifcruz/CyberEther #20: Missing license information

mikeryan, 2023-04-18:
Hey, super cool project! It would be awesome if you could add a LICENSE file to it so users/developers/packagers know under what terms it may be distributed. Thanks!

luigifcruz, 2023-06-25:
Hi Mike,

Thanks!

I’m not completely sure which direction to go with licensing for this project yet. Can you elaborate on your use case for this library? Your answer would help me choose the best license possible.

Best Regards,
Luigi Cruz

devnulling, 2023-10-03:
I'll echo what @mikeryan said, this is a great project, it's exciting to see how it has progressed. A license would be useful to help developers know if they can use this project in certain applications. 

My limited opinion of it boils down to if you want to allow use in closed source applications with something such as the MIT/Apache license. If not, GPLv3/LGPLv3 tends to be the popular choice.

luigifcruz, 2023-12-18:
Thanks for chiming in, @devnulling!

I gave a lot of thought to this licensing situation. This was the most difficult decision I had to make throughout the entire project. But I think I finally found a suitable license for the project that will benefit everyone. I just need to tighten some bolts here and there first.

I'll have more news to share in a bit. Maybe by Christmas!

luigifcruz, 2024-01-05:
Thanks all for the input. The licensing is now disclosed!

closed 2024-01-05 by luigifcruz