luigifcruz/CyberEther #22: A section about how to integrate the SDR would be nice.

luigifcruz, 2023-04-25:
A section about how to integrate the SDR would be nice.
I am currently trying to get CyberEther running with a USRP b200 on a mac M1, but am having problems installing the corresponding SoapySDR module (which I understand is required?).
The error is also almost a year old with apparently no responses from the maintainers https://github.com/pothosware/SoapyUHD/issues/50.

I saw that you had CyberEther running with a different UHD device, was there anything special you did?

_Originally posted by @nzqo in https://github.com/luigifcruz/CyberEther/issues/11#issuecomment-1497689865_

luigifcruz, 2023-04-25:
Moving this comment to a new issue.

luigifcruz, 2023-04-25:
@nzqo That's weird. I installed UHD without problems in macOS. SoapySDR should work the same way with CyberEther.

nzqo, 2023-04-25:
> @nzqo That's weird. I installed UHD without problems in macOS. SoapySDR should work the same way with CyberEther.

Did you use brew or build and install manually? I was thinking of debugging this but couldn't reproduce the error when building manually. Seems that one of the brew packages involved is broken

luigifcruz, 2023-04-27:
I installed it manually. I also had previous problems with Soapy packages provided by Brew.

luigifcruz, 2024-01-05:
Please reffer

closed 2024-01-05 by luigifcruz