luigifcruz/CyberEther #25: terminate called after throwing an instance of 'std::runtime_error'   what():  Failed to make connection with ''

higgins-sdr, 2023-06-23:
Hi Luigi,

I installed CyberEther with the new Vulkan capabilities.
When I try to run the program I get the following error:
```
./cyberether
Welcome to CyberEther!
JETSTREAM [DEBUG] | Initializing CPU backend.
JETSTREAM [DEBUG] | Initializing Vulkan backend.
JETSTREAM [DEBUG] | [VULKAN] Required layer found: VK_LAYER_KHRONOS_validation
JETSTREAM [DEBUG] | [VULKAN] Device supports required extension: VK_KHR_swapchain
JETSTREAM [DEBUG] | [VULKAN] Device supports required extension: VK_EXT_memory_budget
JETSTREAM [DEBUG] | [VULKAN] Device supports required extension: VK_KHR_swapchain
JETSTREAM [INFO]  | —————————————————————————————————————————————————————
JETSTREAM [INFO]  | Jetstream Heterogeneous Backend [VULKAN]
JETSTREAM [INFO]  | —————————————————————————————————————————————————————
JETSTREAM [INFO]  | Device Name:     Quadro P600
JETSTREAM [INFO]  | Device Type:     DISCRETE
JETSTREAM [INFO]  | API Version:     1.3.205
JETSTREAM [INFO]  | Unified Memory:  YES
JETSTREAM [INFO]  | Processor Count: 8
JETSTREAM [INFO]  | Physical Memory: 2.24 GB
JETSTREAM [INFO]  | Staging Buffer:  32.00 MB
JETSTREAM [INFO]  | —————————————————————————————————————————————————————
JETSTREAM [DEBUG] | Creating Linux viewport.
JETSTREAM [DEBUG] | Window initialized.
JETSTREAM [DEBUG] | Initializing Soapy module.
JETSTREAM [DEBUG] | Initializing Window module.
JETSTREAM [DEBUG] | Initializing Filter module.
JETSTREAM [DEBUG] | Initializing Multiply module.
JETSTREAM [DEBUG] | Initializing FFT module.
JETSTREAM [DEBUG] | Initializing Multiply module.
JETSTREAM [DEBUG] | Initializing Amplitude module.
JETSTREAM [DEBUG] | Initializing Scale module.
JETSTREAM [DEBUG] | Initializing FFT module.
JETSTREAM [DEBUG] | Initializing Lineplot module.
JETSTREAM [DEBUG] | Initializing Waterfall module.
JETSTREAM [DEBUG] | Initializing Spectrogram module.
JETSTREAM [DEBUG] | Initializing Constellation module.
JETSTREAM [DEBUG] | Creating compute graph.
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | [000] [Soapy] [Device::CPU] [C: YES, P: NO]
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | Configuration:
JETSTREAM [INFO]  |      Device String:      driver=lime
JETSTREAM [INFO]  |      Frequency:          96.90 MHz
JETSTREAM [INFO]  |      Sample Rate:        10.00 MHz
JETSTREAM [INFO]  |      Buffer Multiplier:  512
JETSTREAM [INFO]  |      Batch Size:         16
JETSTREAM [INFO]  |      Output Buffer Size: 2048
JETSTREAM [INFO]  | Block I/O:
JETSTREAM [INFO]  |     Inputs:
JETSTREAM [INFO]  |         None
JETSTREAM [INFO]  |     Outputs:
JETSTREAM [INFO]  |         0: [CF32] [16, 2048] | [Device::CPU] | Pointer: 0x55a0a13e8000 | Hash: 0x2ce4099f35ad28a4 | Pos: 0
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | [001] [Window] [Device::CPU] [C: NO, P: NO]
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | Configuration:
JETSTREAM [INFO]  |     Window Shape: [16, 2048]
JETSTREAM [INFO]  | Block I/O:
JETSTREAM [INFO]  |     Inputs:
JETSTREAM [INFO]  |         None
JETSTREAM [INFO]  |     Outputs:
JETSTREAM [INFO]  |         0: [CF32] [16, 2048] | [Device::CPU] | Pointer: 0x55a0a1429000 | Hash: 0xd5a4b3ecc2b72f47 | Pos: 0
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | [002] [Filter] [Device::CPU] [C: NO, P: NO]
JETSTREAM [INFO]  | ——————————————————————————————————————————————————————————————————————————————————————————————————————————
JETSTREAM [INFO]  | Configuration:
JETSTREAM [INFO]  |     Signal Sample Rate: 10000000
[... 177 lines cut ...]
JETSTREAM [DEBUG] | Program initialized.
JETSTREAM [DEBUG] | Texture initialized.
JETSTREAM [DEBUG] | Surface initialized.
JETSTREAM [DEBUG] | [VULKAN] Binding surface to window.
JETSTREAM [DEBUG] | Buffer initialized.
JETSTREAM [DEBUG] | Vertex initialized.
JETSTREAM [DEBUG] | Draw initialized.
JETSTREAM [DEBUG] | Buffer initialized.
JETSTREAM [DEBUG] | Vertex initialized.
JETSTREAM [DEBUG] | Draw initialized.
JETSTREAM [DEBUG] | Texture initialized.
JETSTREAM [DEBUG] | Program initialized.
JETSTREAM [DEBUG] | Program initialized.
JETSTREAM [DEBUG] | Texture initialized.
JETSTREAM [DEBUG] | Surface initialized.
JETSTREAM [DEBUG] | [VULKAN] Binding surface to window.
[INFO] Make connection: ''
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to make connection with ''
Aborted (core dumped)
```
Any pointers regarding this error?

Best,
J

higgins-sdr, 2023-07-03:
@luigifcruz thanks for the pointer.
I also noticed the webgpu [branch](https://github.com/luigifcruz/CyberEther/pull/27). Is it ready for trying it out? And are there any installation instructions for that branch? 

I also noticed that the `server.py` sets up the CORS headers (why is that required?).

luigifcruz, 2023-07-05:
At this moment, the WebGPU branch doesn't work. It is under active development and should be ready in the coming weeks.

The Python server is necessary for testing. The CORS headers are needed because the application uses threads and uses SharedArrays to pass data around. Due to security concerns from the Google Chrome team, the access to SharedArrays were limited and now require CORS to be enabled.

higgins-sdr, 2023-07-06:
Ok that's what I thought as well. So you are using SharedArrayBuffers for sharing the data from the main thread to the worker threads, right? Do you run each component (line plot, spectrogram, eye plot etc) in its own thread? Also are you explicitly using web workers within javascript or are you delegating the handling of the web-workers to whatever wasm implementation you have?

Thanks once again for your answer, this is a very exciting project to follow.

luigifcruz, 2023-07-24:
@higgins-sdr Sorry for the delay. Yes, I think that's how Emscripten implemented `pthreads` behind the scenes. No, currently all graphic components run in a single thread. There is little to no advantage in running them in different threads most of the time. Since I'm targeting a portable codebase, the Web Workers instantiation is offloaded to Emscripten. All threads are implemented using `std::thread`.

Thanks! The WebGPU backend is now working. The current implementation needs a patched Emscripen to run, but I'm planning to remove this restriction in the coming weeks.