luigifcruz/CyberEther #33: Vulkan window crahses when resized in windows

Paulo-D2000, 2023-09-22:
Resizing the window causes the program to crash on windows x64

Visual Studio says its on DIspatchCmdBeginRenderPass
[image]

Paulo-D2000, 2023-09-22:
Oh, wait... it only crashes if it has one flowgraph running it seems like...

Paulo-D2000, 2023-09-22:
Update again... I think its the sopay rtlsdr thats buggint everything... with the pluto i think it works

luigifcruz, 2023-10-04:
Oh, thanks for investigating this! Please, re-open this if actually related to the graphical backend. The problem with RTL-SDR should be fixed once `v1.0` is released.

closed 2023-10-04 by luigifcruz

luigifcruz, 2023-10-14:
I found a bug in the Vulkan code that might caused this issue. d4bb62246e6fc55fd19101c7f69c919679b8afd7