luigifcruz/CyberEther #35: Unable to build on macOS

erikt, 2023-09-23:
I am unable to get CyberEther to build on macOS. The documentation is sparse, but following the `INSTALL.md` results in build seemingly stalling with this output:
```
↳ ninja -j`nproc`
[23/87] Generating subprojects/jetstream/shaders/waterfall_signal.frag.wgsl with a custom command	
```

macOS:
```
↳ sw_vers 
ProductName:		macOS
ProductVersion:		14.0
BuildVersion:		23A344
```

clang:
```
↳ clang --version
Apple clang version 15.0.0 (clang-1500.0.40.1)
Target: arm64-apple-darwin23.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Xcode:
```
↳ xcodebuild -version 
Xcode 15.0
Build version 15A240d
```

Hardware:
```
↳ system_profiler SPHardwareDataType 
Hardware:

    Hardware Overview:

      Model Name: Mac mini
      Model Identifier: Macmini9,1
      Model Number: MGNT3KS/A
      Chip: Apple M1
      Total Number of Cores: 8 (4 performance and 4 efficiency)
      Memory: 8 GB
      System Firmware Version: 10151.1.1
      OS Loader Version: 10151.1.1
```

meson output:
```
↳ HERE=`pwd`
~/Developer/3rd_party/CyberEther (main)
↳ meson setup $HERE $HERE/build
The Meson build system
Version: 1.2.1
Source dir: /Users/erik/Developer/3rd_party/CyberEther
Build dir: /Users/erik/Developer/3rd_party/CyberEther/build
Build type: native build
Project name: cyberether
Project version: 0.0.0
C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.0.40.1)")
C++ linker for the host machine: c++ ld64 1015.7
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency threads found: YES

Executing subproject jetstream 

jetstream| Project name: jetstream
jetstream| Project version: 0.1.0
jetstream| C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.0.40.1)")
jetstream| C++ linker for the host machine: c++ ld64 1015.7
jetstream| Dependency threads found: YES unknown (cached)
jetstream| Found pkg-config: /opt/homebrew/bin/pkg-config (0.29.2)
jetstream| Run-time dependency fmt found: YES 10.1.0
jetstream| Run-time dependency CUDA (modules: cudart, dl) found: NO (tried system)
jetstream| Run-time dependency fftw3f found: YES 3.3.10
jetstream| Run-time dependency appleframeworks found: YES (MetalKit, Metal, Foundation)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program spirv-cross found: YES (/opt/homebrew/bin/spirv-cross)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Run-time dependency glfw3 found: YES 3.3.8
jetstream| Run-time dependency soapysdr found: YES 0.8.1
jetstream| Run-time dependency vulkan found: NO (tried pkgconfig and system)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program spirv-cross found: YES (/opt/homebrew/bin/spirv-cross)
jetstream| Program naga found: YES (/opt/homebrew/bin/naga)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Objective-C++ compiler for the host machine: c++ (clang 15.0.0)
jetstream| Objective-C++ linker for the host machine: c++ ld64 1015.7
jetstream| Configuring jetstream_config.hh using configuration
jetstream| Build targets in project: 36
jetstream| Subproject jetstream finished.

Build targets in project: 37

jetstream 0.1.0

  Loaders
    CUDA      : NO
    FFTW      : YES
    Metal     : YES
    GLFW      : YES
    SoapySDR  : YES
    Vulkan    : NO
    WebGPU    : NO

  Backend (Compute)
    CPU       : YES
    CUDA      : NO
    Metal     : YES
    Vulkan    : NO
    WebGPU    : NO

  Viewport
    GLFW      : YES
    iOS       : NO

  Render (Present)
    Metal     : YES
    Vulkan    : NO
    WebGPU    : NO

  FFT Module
    CPU (FFTW): YES
    Metal     : YES

  Window Module
    CPU       : YES
    Metal     : YES

  Multiply Module
    CPU       : YES
    Metal     : YES

  Amplitude Module
    CPU       : YES
    Metal     : YES

  Scale Module
    CPU       : YES
    Metal     : YES

  Filter Module
    CPU       : YES
    Metal     : YES

  Soapy Module
    CPU       : YES
    Metal     : YES

  Lineplot Module
    CPU       : YES
    Metal     : YES

  Waterfall Module
    CPU       : YES
    Metal     : YES

  Spectrogram Module
    CPU       : YES
    Metal     : YES

  Constellation Module
    CPU       : YES

  Graph Backend
    CPU       : YES
    Metal     : YES

  General
    cpp_std   : c++20
    prefix    : /opt/homebrew
    buildtype : debug
    crossbuild: NO

cyberether 0.0.0

  General
    cpp_std   : c++20
    prefix    : /opt/homebrew
    buildtype : debug
    crossbuild: NO

  Subprojects
    jetstream : YES

Found ninja-1.11.1 at /opt/homebrew/bin/ninja
```

luigifcruz, 2023-09-23:
That's weird. Never saw the build process hanging and I can't find anything wrong with the logs. Can you try to compile the `development` branch instead?

erikt, 2023-09-23:
Same result.

erikt, 2023-09-23:
Same result.

Trying to debug the issue, I ran what seems like the command where it hangs:
```
↳ /opt/homebrew/bin/glslangValidator -V -S vert ../shaders/spectrogram/signal.vert.glsl -o shaders/spectrogram_signal.vert.spv 
../shaders/spectrogram/signal.vert.glsl
```
Seems to work.

luigifcruz, 2023-09-23:
Are any WGSL files being generated at `build/shaders/`? Maybe `naga` is hanging.

erikt, 2023-09-23:
```
↳ tree shaders 
shaders
├── constellation
├── constellation_shaders.hh
├── constellation_signal.frag.msl
├── constellation_signal.frag.spv
├── constellation_signal.vert.msl
├── constellation_signal.vert.spv
├── lineplot
├── lineplot_grid.frag.msl
├── lineplot_grid.frag.spv
├── lineplot_grid.vert.msl
├── lineplot_grid.vert.spv
├── lineplot_shaders.hh
├── lineplot_signal.frag.msl
├── lineplot_signal.frag.spv
├── lineplot_signal.vert.msl
├── lineplot_signal.vert.spv
├── spectrogram
├── spectrogram_signal.frag.msl
├── spectrogram_signal.frag.spv
├── spectrogram_signal.vert.msl
├── spectrogram_signal.vert.spv
└── waterfall

5 directories, 18 files
```

erikt, 2023-09-23:
What's `naga`?

luigifcruz, 2023-09-23:
[Naga](https://github.com/gfx-rs/naga) is a shader translation tool  for WebGPU. Looks like this is your problem. Try uninstalling it with Brew (`brew uninstall naga-cli`) and running the compilation again.

I'm not sure if I understand the root cause but this should solve your problems.

erikt, 2023-09-23:
Ah. I actually think the problem was that `naga-cli` was not installed. I installed it with brew and the build works. Thanks for the help.

closed 2023-09-23 by erikt

luigifcruz, 2023-10-05:
For completeness, this bug was fixed in 4b9350d217e1834838aa20aa8a70ef50b86d3530.