luigifcruz/CyberEther #38: Value "c++20" (of type "string") for combo option "C++ language standard to use" is not one of the choices.

audioplatinum, 2023-09-29:
I got this error when compiling, how we can skip this? 

`subprojects/jetstream/src/render/metal/tools/meson.build:1:0: ERROR: Value "c++20" (of type "string") for combo option "C++ language standard to use" is not one of the choices. Possible choices are (as string): "none", "c++98", "c++11", "c++14", "c++17", "gnu++98", "gnu++11", "gnu++14", "gnu++17".`

c++ version:
`jetstream| C++ compiler for the host machine: ccache c++ (clang 12.0.5 "Apple clang version 12.0.5 (clang-1205.0.22.9)")`

ion-concepts, 2023-10-02:
Ran into that last night and it seemed to be a `meson` issue after some googling. 
I had installed a venv to build this (OSX 13.4) and pip installed `meson` which gave me version 1.2.2
This gave me the same issue.

Then whilst debugging I accidentally ran ` meson setup $HERE $HERE/build` outside the venv and it picked up my homebrew installed meson at version 1.2.99 and ran to completion with no issues.

Still stuck with another issue but hope this helps!

ion-concepts, 2023-10-02:
Also, the meson issue that gave me the clue: mesonbuild/meson#11890

luigifcruz, 2023-10-04:
This is a very stupid error caused by Meson. Their Objective-C++ compiler uses hard-coded standards codes and they never updated them with the C++20 standard. I reported this [error early](https://github.com/mesonbuild/meson/pull/11461) this year. However, the Meson team is being very slow in releasing a stable version containing the fix linked above by @ion-concepts.

luigifcruz, 2023-10-04:
I asked the Meson team to consider distributing the patch as a minor release instead of version `1.3.0`. Meanwhile, as @ion-concepts suggested, this can be avoided by installing the head version of Meson with Brew `brew install --HEAD meson`. Make sure to uninstall any previously installed Meson version (e.g. `pip uninstall meson`).

luigifcruz, 2024-01-05:
The first alpha release is now live. This issue should be resolved with the latest version of Meson.

closed 2024-01-05 by luigifcruz