luigifcruz/CyberEther #39: ninja never completes, build seems to loop on syscalls.

ion-concepts, 2023-10-02:
CyberEther@a6e9e08a0a8dce4385b606e585500c09e3603a1e
OSX 13.4.1, meson 1.2.99, ninja 1.11.1

Meson seems to configure build fine but ninja seems to loop indefinitely burning CPU in system calls (See attachments)

[image]
[image]
[image]

```ianb@Muonator CyberEther % meson setup $HERE $HERE/build
The Meson build system
Version: 1.2.99
Source dir: /Users/ianb/src/luigifcruz/CyberEther
Build dir: /Users/ianb/src/luigifcruz/CyberEther/build
Build type: native build
Project name: cyberether
Project version: 0.0.0
C++ compiler for the host machine: c++ (clang 14.0.3 "Apple clang version 14.0.3 (clang-1403.0.22.14.1)")
C++ linker for the host machine: c++ ld64 857.1
Host machine cpu family: aarch64
Host machine cpu: aarch64
Run-time dependency threads found: YES

Executing subproject jetstream

jetstream| Project name: jetstream
jetstream| Project version: 0.1.0
jetstream| C++ compiler for the host machine: c++ (clang 14.0.3 "Apple clang version 14.0.3 (clang-1403.0.22.14.1)")
jetstream| C++ linker for the host machine: c++ ld64 857.1
jetstream| Dependency threads found: YES unknown (cached)
jetstream| Found pkg-config: YES (/opt/homebrew/bin/pkg-config) 0.29.2
jetstream| Run-time dependency fmt found: YES 10.1.0
jetstream| Run-time dependency CUDA (modules: cudart_static, dl) found: NO (tried system)
jetstream| Run-time dependency fftw3f found: YES 3.3.10
jetstream| Run-time dependency appleframeworks found: YES (MetalKit, Metal, Foundation)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program spirv-cross found: YES (/opt/homebrew/bin/spirv-cross)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Run-time dependency glfw3 found: YES 3.3.8
jetstream| Run-time dependency soapysdr found: YES 0.8.1
jetstream| Run-time dependency vulkan found: YES 1.3.263
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Program glslangValidator found: YES (/opt/homebrew/bin/glslangValidator)
jetstream| Program spirv-cross found: YES (/opt/homebrew/bin/spirv-cross)
jetstream| Program naga found: YES (/opt/homebrew/bin/naga)
jetstream| Program python3 found: YES (/opt/homebrew/bin/python3)
jetstream| Objective-C++ compiler for the host machine: c++ (clang 14.0.3)
jetstream| Objective-C++ linker for the host machine: c++ ld64 857.1
jetstream| Configuring jetstream_config.hh using configuration
jetstream| Build targets in project: 36
jetstream| Subproject jetstream finished.

Build targets in project: 37```

luigifcruz, 2023-10-04:
Thanks for reporting this bug! 

Looks like this is a multifactorial bug. For it to happen two things need to happen:
1. Brew incorrectly reports [`naga`](https://github.com/gfx-rs/naga) is installed.
2. Python script, for some reason, busy-looping into oblivion trying to execute something non-existent.

Naga is required by CyberEther to convert the GLSL shaders to the WebGPU Shading Language. I'm working on fixing these two failure points, but in the meantime, the absolute minimal fix for this would be installing the [naga](https://formulae.brew.sh/formula/naga-cli#default) package with brew.

luigifcruz, 2023-10-04:
I'm actually impressed by how random these errors had to be to create this multi-threaded busy loop. 😄

luigifcruz, 2023-10-05:
Oh, the plot thickens. This is likely due to the [snake game named Naga](https://github.com/anayjoshi/naga/) being installed instead of the [shader translator named Naga](https://github.com/gfx-rs/naga). The busy loop is caused by the game Naga running on a virtual terminal session.

I initially tried to fix this conflict by checking the version of the Naga binary installed. But the game Naga doesn't respond to the `--version` command. I'm thinking about another alternative.

ion-concepts, 2023-10-05:
Ha! This starts to make sense...I double checked I had a homebrew `naga` install earlier today, then scratched my head!!

luigifcruz, 2023-10-05:
Done! The commit above should fix this problem. It compiles Naga from the source automatically instead of relying on the script installed by Brew. This looks a bit overkill but is fine considering no package manager on Linux has Naga, so this had to be done anyway. Let me know if this works for ya!

luigifcruz, 2024-01-05:
The first alpha release is now live. This issue should be resolved by now. Please re-open otherwise. Thanks!

closed 2024-01-05 by luigifcruz