luigifcruz/CyberEther #45: File dialog for Windows platform

luigifcruz, 2024-01-06:
We currently support file picker dialog and save file dialog for macOS, Linux, and Browser. We need the same thing but for Windows. 

PS: No external dependencies should be added for this.

Paulo-D2000, 2024-01-07:
I've added the dialogs on the code... I'ts been a long time since I worked with meson, how do I set the linker libraries ?

luigifcruz, 2024-01-07:
Which libraries do you need? System ones?

Paulo-D2000, 2024-01-07:
Yeah, comsuppw.lib, Comdlg32.lib, kernel32.lib, Shell32.lib

Paulo-D2000, 2024-01-07:
I'm downloading the build tools here so I haven't tested it yet... (Also, we could add one github CI action to make the windows built too XD)
https://github.com/Paulo-D2000/CyberEther/commit/daf9a8dbd26173f5c51fc56918bd014e4743520a

luigifcruz, 2024-01-07:
The draft looks good. Yeah, I'm planning to add that too.

Paulo-D2000, 2024-01-07:
Hm, stange.... running meson with mingw-w64 vulkan doesn't get detected... only with msvc :\

luigifcruz, 2024-01-07:
Is the VULKAN_SDK environmental variable in PATH?

Paulo-D2000, 2024-01-07:
Yeah, Not even MSVC is detecting it now... LOL
[image]

luigifcruz, 2024-01-07:
Maybe the installation got corrupt. Try reinstalling it.

Paulo-D2000, 2024-01-07:
Oh, there was another VULKAN_SDK var set to C.. My mistake, LOL

Paulo-D2000, 2024-01-07:
Hmm.. its still giving the error "Dependency lookup for vulkan with method 'system' failed: VULKAN_SDK point to invalid directory (no lib)" on the log... I've reinstalled vulkan 2 times...

luigifcruz, 2024-01-07:
Oh, are you setting it up as a static library? 

```
meson build-static --default-library=static build-static
```

Paulo-D2000, 2024-01-07:
This error is on the setup command on meson :/ If I try running the command you suggested I get `` meson: error: unrecognized arguments: build-static``

luigifcruz, 2024-01-07:
Sorry, I added an extra `build-static` at the end. Try `meson build-static --default-library=static`.

Paulo-D2000, 2024-01-07:
Idk why, but with Mingw it simply doesn't detect any libs, not even soapysdr... LOL, Maybe its because the libs are compiled with msvc usually ?

luigifcruz, 2024-01-07:
No, this is expected. Other than Vulkan, everything should be compiled internally by Meson. Including SoapySDR, GLFW, etc.

Paulo-D2000, 2024-01-07:
Oh, i think I've installed 32bit mingw...

Paulo-D2000, 2024-01-07:
yeah, i868, lol

luigifcruz, 2024-01-07:
💀

Paulo-D2000, 2024-01-07:
LOL now its using my ARM GCC haha

luigifcruz, 2024-01-07:
I believe I used [this](https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/installer/mingw-w64-install.exe) online installer to install MinGW-w64.

Paulo-D2000, 2024-01-07:
Yeah, I have it installed but both gcc arm and gcc x86_64 are on the windows path... I've set cc and cxx vars so it works now... comsuppw.lib is msvc only (compiler function to convert strings) Let me fix it...

Paulo-D2000, 2024-01-07:
70/198... lets see...

Paulo-D2000, 2024-01-07:
If I run ninja install, where it will be installed ? Can it just be run from the build folder (static so I guess yes, right?)

luigifcruz, 2024-01-07:
Yes. The output binary should be portable. Except for Vulkan.

Paulo-D2000, 2024-01-07:
Build is Almost done... (If it works)  might be interesting to add the function [CommDlgExtendedError](https://learn.microsoft.com/en-us/windows/desktop/api/commdlg/nf-commdlg-commdlgextendederror) to get more detaisl about errors / exceptions opening / saving files... The functions return True or false only...

Paulo-D2000, 2024-01-07:
Windows macros attack again...
[image]

luigifcruz, 2024-01-07:
The procedure to import `windows.h` is this one:

```
#define WIN32_LEAN_AND_MEAN
#include <windows.h>
#undef ERROR
#undef FATAL
```

Seems to be working for now.

luigifcruz, 2024-01-07:
> might be interesting to add the function [CommDlgExtendedError](https://learn.microsoft.com/en-us/windows/desktop/api/commdlg/nf-commdlg-commdlgextendederror) to get more details about errors / exceptions opening / saving files... The Open and Save functions return True or false only...

I considered doing something like this, but I think a binary status is good enough for now. Further details can be shown to the user with `JST_ERROR`.

Paulo-D2000, 2024-01-07:
it builded ok... lets test...

Paulo-D2000, 2024-01-07:
#define WIN32_LEAN_AND_MEAN caused errors because I'm using one macro OPENFILENAMEA...

Paulo-D2000, 2024-01-07:
oh, entry point not found... LOL

Paulo-D2000, 2024-01-07:
Trying a clean rebuild yielded gcc errors... lol...
```
C:/MinGW/lib/gcc/x86_64-w64-mingw32/13.2.0/include/c++/bits/std_thread.h: In function 'bool std::operator==(thread::id, thread::id)':
C:/MinGW/lib/gcc/x86_64-w64-mingw32/13.2.0/include/c++/bits/std_thread.h:330:26: error: no match for 'operator==' (operand types are 'std::thread::native_handle_type' and 'std::thread::native_handle_type')
  330 |     return __x._M_thread == __y._M_thread;
      |            ~~~~~~~~~~~~~ ^~ ~~~~~~~~~~~~~
      |                |                |
      |                |                std::thread::native_handle_type
      |                std::thread::native_handle_type```

Paulo-D2000, 2024-01-07:
I've ''hacked'' mingw following http://hectorhon.blogspot.com/2018/05/building-libpqxx-on-msys2-mingw-64-bit.html and now i get the errors...
[image]

luigifcruz, 2024-01-07:
Does it compile without your patches? It compiled fine on my machine. If not, did you install MinGW-w64 with `posix` thread instead of `min32`?

Paulo-D2000, 2024-01-07:
Yeah, my gcc is posix, should it be win32 ?

luigifcruz, 2024-01-07:
No, it should be `posix`. Try to compile without the patches. If it works, we know something is up with the dependencies.

Paulo-D2000, 2024-01-07:
Nope... It doesn't work on the ''main'' version without patches either... The error seems to be like on Soapy Airspy that is using multithreading...

luigifcruz, 2024-01-07:
Weird the main worked on Windows 11. It might be something wrong with your MinGW-w64 installation.

Paulo-D2000, 2024-01-07:
The mingw installer on your link is for gcc 8.1... I'm using gcc 13.2

Paulo-D2000, 2024-01-07:
I'm using this mingw... [x86_64-13.2.0-release-posix-seh-ucrt-rt_v11-rev0.7z](https://github.com/niXman/mingw-builds-binaries/releases/download/13.2.0-rt_v11-rev0/x86_64-13.2.0-release-posix-seh-ucrt-rt_v11-rev0.7z)

Paulo-D2000, 2024-01-07:
Maybe ucrt has some bugs... I will try with the older runtime...

luigifcruz, 2024-01-07:
I think the installer will download the latest version available. But I cannot verify because I don't have access to my computer right now. I would suggest installing it using the installer since I don't completely know if I used UCRT or MSVCRT.

Paulo-D2000, 2024-01-07:
Hmm the installer gives error... Does gcc 8.10 even support cpp20 ?

luigifcruz, 2024-01-07:
Are you sure it's GCC 8? This would have not worked for me if it had been.

Paulo-D2000, 2024-01-07:
It tries to download 8.10... Maybe you downloaded mingw from another link 
[image]
[image]

luigifcruz, 2024-01-07:
Yeah, that's not the installer I used. Please, try if the MSVCRT version works. 

https://github.com/niXman/mingw-builds-binaries/releases/download/13.2.0-rt_v11-rev1/x86_64-13.2.0-release-posix-seh-msvcrt-rt_v11-rev1.7z

Paulo-D2000, 2024-01-07:
Nope, I tried msvcrt too, same error... Maybe its the static build ? or some config on the airspy lib ?

luigifcruz, 2024-01-07:
Both unlikely, this worked here. Try removing the soapyairspy as a dependency.

Paulo-D2000, 2024-01-07:
Nope... if I remove airspy then rtlsdr shows the same error... removing it I get the macro redefined errors that happened on the ''normal'' build...

Paulo-D2000, 2024-01-07:
Its strange because I didn't find any #undefs on your code... when I tryed the windows build I needed to add the #undefs on the types.hh file and other files that used max, min, etc

luigifcruz, 2024-01-07:
Something is probably wrong with your installation then. Maybe worth checking if there isn't a conflict between different MinGW-w64 versions. Or an old version of a header from a previous CyberEther install.

Paulo-D2000, 2024-01-07:
Hmm, It could be because I'm running the build on the gnuradio env... what deps I need to build it from a ''clean'' system ?

luigifcruz, 2024-01-07:
Vulkan

Paulo-D2000, 2024-01-07:
hmmm, i think it was the gnuradio env... It was using pthreads and other headers from there instead of gcc... compiling now 147/198

Paulo-D2000, 2024-01-07:
Ok... It builded... but throwed the string entry point error again... (without changes)

Paulo-D2000, 2024-01-07:
hmm I copyied libstc++.dll to the exe location and it worked!!

luigifcruz, 2024-01-07:
Maybe it's linking with something? It shouldn't like dynamically with anything but Vulkan and system libraries.

Paulo-D2000, 2024-01-07:
It works!! (But the save needs to add the default exension perhaps ?)
[image]

[image]

Paulo-D2000, 2024-01-07:
> Maybe it's linking with something? It shouldn't like dynamically with anything but Vulkan and system libraries.

I think its missing -static-libstdc++ flag perhaps... its using the dynamic lib...

[5 more replies]

closed 2024-01-13 by luigifcruz