luigifcruz/CyberEther #51: `libfmt.so.9` not found after compiling

patriciogonzalezvivo, 2024-01-07:
I'm on Ubuntu 22.04, followed the compiling instructions with no error.
When trying to run `cyberether` reports:
```bash
[~/Desktop/CyberEther/build]$ cyberether
cyberether: error while loading shared libraries: libfmt.so.9: cannot open shared object file: No such file or directory
```
I try installed `libfmt-dev` through `apt` with no luck

luigifcruz, 2024-01-07:
Can you send me the Meson logs (`CyberEther/build/meson-logs/meson-log.txt`) for this build? You shouldn't need to have `fmtlib` installed for CyberEther.

patriciogonzalezvivo, 2024-01-17:
CyberEther/build/meson-logs/meson-log.txt

```
Build started at 2024-01-17T02:55:45.707557
Main binary: /usr/bin/python3
Build Options: -Dbuildtype=debugoptimized
Python system: Linux
The Meson build system
Version: 0.61.2
Source dir: /home/patricio/Desktop/CyberEther
Build dir: /home/patricio/Desktop/CyberEther/build
Build type: native build
Project name: CyberEther
Project version: 1.0.0
Sanity testing C++ compiler: c++
Is cross compiler: False.
Sanity check compiler command line: c++ sanitycheckcpp.cc -o sanitycheckcpp.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/patricio/Desktop/CyberEther/build/meson-private/sanitycheckcpp.exe
C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Sanity testing C++ compiler: c++
Is cross compiler: False.
Sanity check compiler command line: c++ sanitycheckcpp.cc -o sanitycheckcpp.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/patricio/Desktop/CyberEther/build/meson-private/sanitycheckcpp.exe
C++ compiler for the build machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the build machine: c++ ld.bfd 2.38
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: x86_64
Run-time dependency threads found: YES
Pkg-config binary for 1 is not cached.
Pkg-config binary missing from cross or native file, or env var undefined.
Trying a default Pkg-config fallback at pkg-config
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Determining dependency 'fmt' with pkg-config executable '/usr/bin/pkg-config'
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --modversion fmt` -> 0
8.1.1
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --cflags fmt` -> 0

env[PKG_CONFIG_ALLOW_SYSTEM_LIBS]: 1
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --libs fmt` -> 0
-L/usr/lib/x86_64-linux-gnu -lfmt
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --libs fmt` -> 0
-lfmt
Running compile:
Working directory:  /home/patricio/Desktop/CyberEther/build/meson-private/tmp4gu6it90
Command line:  c++ /home/patricio/Desktop/CyberEther/build/meson-private/tmp4gu6it90/testfile.cpp -o /home/patricio/Desktop/CyberEther/build/meson-private/tmp4gu6it90/output.exe -D_FILE_OFFSET_BITS=64 -O0 -fpermissive 

Code:
 #include<stdio.h>
        
[... 424 lines cut ...]
  General
    cpp_std          : c++20
    prefix           : /usr/local
    buildtype        : debugoptimized
    crossbuild       : NO

  Subprojects
    naga             : YES

  User defined options
    buildtype        : debugoptimized

Found ninja-1.10.2.git.kitware.jobserver-1 at /usr/local/bin/ninja
Running compile:
Working directory:  /tmp/tmpfy9yuw4t
Command line:  c++ /tmp/tmpfy9yuw4t/testfile.cpp -o /tmp/tmpfy9yuw4t/output.obj -D_FILE_OFFSET_BITS=64 -c -std=c++20 

Code:
 int i = static_cast<int>(0);
Compiler stdout:
 
Compiler stderr:
 
Compiler accepts -std=c++20: YES
```

patriciogonzalezvivo, 2024-01-17:
I'm using Ubuntu 22.04 in a laptop with a RTX3080. It's weird because I do have CUDA installed

luigifcruz, 2024-01-25:
Thanks! I removed external `fmtlib` from the project. The patch is now live in the `v1.0.0-alpha2` branch that will be released very soon!

luigifcruz, 2024-01-29:
The `v1.0.0-alpha2` is now upstreamed. Let me know if that works for you! Please, make sure to reconfigure the build folder after the pull.

```
$ rm -r build 
$ meson setup -Dbuildtype=debugoptimized build
$ cd build
$ ninja
```

patriciogonzalezvivo, 2024-02-01:
It did worked! thank you!

closed 2024-02-01 by patriciogonzalezvivo