luigifcruz/CyberEther #52: Real to Complex FFT (Feature /Enhacement)

Paulo-D2000, 2024-01-07:
Would be nice to have one Real -> Complex FFT option on the FFT module... It could help debug some of the modules later visualizing the output on the freq. domain...

luigifcruz, 2024-01-07:
Agreed! This is mostly kinda blocked by #48 at this moment. It would be nice to solve that first and then add more modules. But nothing blocks doing this now for CPU.

Paulo-D2000, 2024-01-07:
Ok! I'm trying to make it work here...

Paulo-D2000, 2024-01-07:
It builded... but the flowgraph didn't find any blocks with F32 input... I remeber that on the old version there was one big header with the block input / output definitions / instantiations... Where is it located now ?

luigifcruz, 2024-01-07:
That was automated away because of your feedback back then. But you still need to declare the types in the header [here](https://github.com/luigifcruz/CyberEther/blob/1bb1bbbbc596dae50cbf1f22939f372fdd42ffa7/include/jetstream/modules/fft.hh#L29-L33). The header will also need to be expanded to `..., typename IT, typename OT>`.

Paulo-D2000, 2024-01-07:
I used only T and set the output buffer as CF32 constant... Using IT, OT caused too many bugs :\

Paulo-D2000, 2024-01-07:
Ok, I'm trying with IT, OT on the header... There were some bugs abou expanding the initializer list

Paulo-D2000, 2024-01-07:
Build log... 
[log.txt](https://github.com/luigifcruz/CyberEther/files/13855115/log.txt)

fft.h screenshot...
[image]

luigifcruz, 2024-01-07:
All the templates should have the same number of types. Line 30 should be `MACRO(FFT, CPU, CF32, CF32) \` and so on. Line 36 should be `..., typename IT, typename OT>` without default elements. I would suggest using a module that has two template types as a reference for this. Like the `Cast` module.

Paulo-D2000, 2024-01-07:
Hmmm, on the fft/cpu/base.cc file how would I instantiate the 2 different calls to compute (using pocketfft::c2c or r2c)?

luigifcruz, 2024-01-07:
```cpp
Result FFT<Device::CPU, CF32, CF32>::compute(const RuntimeMetadata&) {  // c2c
Result FFT<Device::CPU, F32, CF32>::compute(const RuntimeMetadata&) { // r2c
```

Paulo-D2000, 2024-01-07:
Hm... Disabling the benchmark it does build... but the module doesn't show on the UI... Trying a clean rebuild...

luigifcruz, 2024-01-07:
You also need to change the block for the FFT to support two types. https://github.com/luigifcruz/CyberEther/blob/1bb1bbbbc596dae50cbf1f22939f372fdd42ffa7/include/jetstream/blocks/fft.hh#L148-L149

Paulo-D2000, 2024-01-08:
It works!!
[image]

Paulo-D2000, 2024-01-08:
Hmmm... Ideally We would ''shift'' the signal (or discard the negative freqs)...
[image]

Paulo-D2000, 2024-01-08:
Unpad did the trick! Amazing! haha
[image]

Paulo-D2000, 2024-01-08:
(Offtopic) About the ''Fast'' FM demod using the central difference and delay...  You can see the attenuation from the central difference kernel at the higher freqs... notice the rds has vanished...
[image]

luigifcruz, 2024-01-08:
Awesome! Very nice work! Are you planning to open a PR for this?

Paulo-D2000, 2024-01-08:
> Awesome! Very nice work! Are you planning to open a PR for this?

Yes! I guess it will also add the windows dialog changes right ?

luigifcruz, 2024-01-08:
No, please create a separate PR for this feature.

Paulo-D2000, 2024-01-08:
Ok, when I get home I will upload the changes to another branch maybe and open the pr

luigifcruz, 2024-01-16:
Closed by #55

closed 2024-01-16 by luigifcruz