luigifcruz/CyberEther #54: Improved AGC logic.

luigifcruz, 2024-01-08:
The current implementation of the AGC is sub-optimal. New implementation should retain good parallelization characteristics and correctly implement averaging.

Paulo-D2000, 2024-01-10:
I've had some good results using a simple control loop like:
```
T agc_work(T input,  float agc_target = 1.0f, float agc_gain = 1e-4f , float agc_max = 655635.0f):
T output = input * gain;
float error = agc_target - std::abs(T); // absolute magnitude of signal
gain += error * agc_gain;
// Optional ? Gain limit...
if(agc_max > 0.0f && gain > agc_max){
  gain = agc_max;
}
```

[WirelessPi AGC](https://wirelesspi.com/how-automatic-gain-control-agc-works/)
[Gnuradio AGC F32](https://github.com/gnuradio/gnuradio/blob/master/gr-analog/include/gnuradio/analog/agc.h#L118)
[Gnuradio AGC CF32](https://github.com/gnuradio/gnuradio/blob/master/gr-analog/include/gnuradio/analog/agc.h#L58)

luigifcruz, 2026-09-14:
Implemented.

closed 2026-09-14 by luigifcruz