luigifcruz/CyberEther #58: not finding RTL-SDR

vielmetti, 2024-01-15:
Hardware: MacBook Air M1

`lsusb`  output

```
> lsusb
Bus 000 Device 001: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838UHIDIR  Serial: 00000001
Bus 000 Device 000: ID 0bda:2838 Realtek Semiconductor Corp. USB 3.1 Bus 
```

version

```
> cyberether --version
CyberEther v1.0.0-debugoptimized
```

Problem: cyberether is not locating this RTL-SDR device, when I use the example "FM decoder" flow graph.

A possible hint is this error?

```
JETSTREAM [ERROR] | Failed to open device. Reason: vector
```

vielmetti, 2024-01-15:
I have the Soapy libraries installed:

```
>       brew install soapysdr soapyrtlsdr
Running `brew update --auto-update`...
Warning: soapysdr 0.8.1_1 is already installed and up-to-date.
To reinstall 0.8.1_1, run:
  brew reinstall soapysdr
Warning: soapyrtlsdr 0.3.3_1 is already installed and up-to-date.
To reinstall 0.3.3_1, run:
  brew reinstall soapyrtlsdr
```

luigifcruz, 2024-01-15:
Can you check if the RTL-SDR module is being listed by `SoapySDRUtil --info`?

vielmetti, 2024-01-15:
```
> SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /opt/local
Search path:  /opt/local/lib/SoapySDR/modules0.8
Module found: /opt/local/lib/SoapySDR/modules0.8/libLMS7Support.so (22.09.0)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

vielmetti, 2024-01-15:
so I think no, based on "Available factories... lime"

luigifcruz, 2024-01-15:
Yeah, it's not being listed. This is a problem with SoapySDR. Maybe reinstalling the module with Brew will fix the problem.

vielmetti, 2024-01-15:
yeah, looks like I might have two copies of SoapySDRInfo on the system. There's on in /opt/local/bin and one in /opt/homebrew/bin -- and it appears that the build I did of the cyberether linked against all of the /opt/local rather than the /opt/homebrew parts.

/opt/homebrew/bin/SoapySDRInfo produces results as expected.

vielmetti, 2024-01-15:
I did a thing that might come back to get me later - copied a module from homebrew into /usr/local/lib - and I'm not proud of that because it's the wrong thing to do but on the other hand it's working now!

Closing, this is not a CyberEther issue.

closed 2024-01-15 by vielmetti