luigifcruz/CyberEther #60: Feature request: squelch

vielmetti, 2024-01-15:
I searched through the codebase and didn't find this, but that doesn't mean it's not here by another name.

The motivation: build a very fast scanner that scans a big swath of the 2 meter ham radio spectrum and plays audio from one or more active connections, but omits signals that are too weak or that have no traffic at the moment.

The filter would take a signal, measure the signal strength, and it it's strong enough it would pass the signal through. if not, then it would omit the data.

luigifcruz, 2024-01-15:
You are right, this doesn't exist yet! But this seems like a very good block to add. I'm adding to the roadmap!

luigifcruz, 2024-01-15:
This block can take a multi-dimensional tensor as input, for example: `{Batch, Channel, Signal}`. The block would detect the power levels for each `Channel` and output the power level signal together with the original input data. The signaling would then be used by a compatible block down the line. For example, the `Sound` block would mute the inactive channels, or the `File Sink` would stop writing data.

Paulo-D2000, 2024-01-17:
This seems interesting... You could make a polyphase channelizer, do the fft, multiply with the input signal fft and do the channel selection on the freq domain peraphs... Then inverse fft to get the channel's output 🤯