luigifcruz/CyberEther #62: cyberether fails: [VULKAN] Error code: VK_ERROR_EXTENSION_NOT_PRESENT

yurivict, 2024-01-25:
The error message should say what exactly is missing:

```
JETSTREAM [DEBUG] | [VULKAN] Creating GLFW viewport.
JETSTREAM [FATAL] | [VULKAN] GLFW error: X11: Vulkan instance missing VK_KHR_xcb_surface extension
JETSTREAM [ERROR] | [VULKAN] GLFW failed to create window surface.
JETSTREAM [ERROR] | [VULKAN] Error code: VK_ERROR_EXTENSION_NOT_PRESENT
Function main (../main.cc@158) throwed!
```

Version: 1.0.0.a1
vulkan-headers-1.3.273         Headers for the Vulkan graphics API
vulkan-loader-1.3.273          Driver loader for the Vulkan graphics API
FreeBSD 14.0

luigifcruz, 2024-01-25:
I think you are the first person to test it in a FreeBSD system. It's probably a system discovery error. 

Can you share the generated `build/includes/jetstream/config.hh` file here? It will help me identify: 1) Why the build system didn't warn you; 2) Verify if this is indeed the case;

yurivict, 2024-01-25:
```
$ cat ./work/CyberEther-1.0.0-alpha1/include/jetstream/backend/config.hh
#ifndef JETSTREAM_BACKEND_CONFIG_HH
#define JETSTREAM_BACKEND_CONFIG_HH

#include "jetstream/logger.hh"
#include "jetstream/types.hh"

namespace Jetstream::Backend {

enum class PhysicalDeviceType : uint8_t {
    UNKNOWN     = 0,
    DISCRETE    = 1,
    INTEGRATED  = 2,
    OTHER       = 3,
};

inline std::ostream& operator<<(std::ostream& os, const PhysicalDeviceType& type) {
    switch (type) {
        case PhysicalDeviceType::UNKNOWN:
            os << "UNKNOWN";
            break;
        case PhysicalDeviceType::DISCRETE:
            os << "DISCRETE";
            break;
        case PhysicalDeviceType::INTEGRATED:
            os << "INTEGRATED";
            break;
        case PhysicalDeviceType::OTHER:
            os << "OTHER";
            break;
        default:
            os.setstate(std::ios_base::failbit);
            break;
    }
    return os;
}

struct Config {
    U64 deviceId = 0;
#ifdef JST_DEBUG_MODE
    bool validationEnabled = true;
#else
    bool validationEnabled = false;
#endif
    U64 stagingBufferSize = 64*1024*1024;
    bool headless = false;
};

}  // namespace Jetstream::Backend

template <> struct fmt::formatter<Jetstream::Backend::PhysicalDeviceType> : ostream_formatter {};

#endif
```

luigifcruz, 2024-01-25:
Sorry, that's another file. I forgot I changed the name of the file. The one you are looking for is called `jetstream_config.hh` inside the build folder include directory.

yurivict, 2024-01-25:
```
$ cat ./work/CyberEther-1.0.0-alpha1/_build/include/jetstream_config.hh
#define JETSTREAM_VERSION_STR "1.0.0"
#define JETSTREAM_BUILD_TYPE "release"
#define JETSTREAM_BUILD_OPTIMIZATION "plain"
#define JETSTREAM_BUILD_DEBUG "False"

#undef JST_IS_STATIC
#define JST_IS_SHARED

#define JST_DEBUG_MODE
#undef JST_RELEASE_MODE

#undef JST_OS_WINDOWS
#undef JST_OS_ANDROID
#undef JST_OS_IOS
#undef JST_OS_MAC
#undef JST_OS_LINUX
#undef JST_OS_BROWSER

// Loaders
#define JETSTREAM_LOADER_CPU_AVAILABLE
/* #undef JETSTREAM_LOADER_CUDA_AVAILABLE */
/* #undef JETSTREAM_LOADER_METAL_AVAILABLE */
#define JETSTREAM_LOADER_GLFW_AVAILABLE
#define JETSTREAM_LOADER_SOAPY_AVAILABLE
/* #undef JETSTREAM_LOADER_WEBGPU_AVAILABLE */
/* #undef JETSTREAM_LOADER_AUDIOTOOLBOX_AVAILABLE */
#define JETSTREAM_LOADER_GSTREAMER_AVAILABLE
/* #undef JETSTREAM_LOADER_WHISPERCPP_AVAILABLE */
// [NEW DEPENDENCY HOOK]

// Backend
#define JETSTREAM_BACKEND_CPU_AVAILABLE
/* #undef JETSTREAM_BACKEND_CUDA_AVAILABLE */
#define JETSTREAM_BACKEND_VULKAN_AVAILABLE
/* #undef JETSTREAM_BACKEND_METAL_AVAILABLE */
/* #undef JETSTREAM_BACKEND_WEBGPU_AVAILABLE */

// Graph
#define JETSTREAM_GRAPH_CPU_AVAILABLE
/* #undef JETSTREAM_GRAPH_METAL_AVAILABLE */
/* #undef JETSTREAM_GRAPH_VULKAN_AVAILABLE */
/* #undef JETSTREAM_GRAPH_WEBGPU_AVAILABLE */

// Render
/* #undef JETSTREAM_RENDER_METAL_AVAILABLE */
#define JETSTREAM_RENDER_VULKAN_AVAILABLE
/* #undef JETSTREAM_RENDER_WEBGPU_AVAILABLE */

// Viewport
#define JETSTREAM_VIEWPORT_HEADLESS_AVAILABLE
#define JETSTREAM_VIEWPORT_GLFW_AVAILABLE
/* #undef JETSTREAM_VIEWPORT_IOS_AVAILABLE */

//
// Modules
//

// FFT
#define JETSTREAM_MODULE_FFT_AVAILABLE
#define JETSTREAM_MODULE_FFT_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_FFT_METAL_AVAILABLE */

// FILTER
#define JETSTREAM_MODULE_FILTER_AVAILABLE
#define JETSTREAM_MODULE_FILTER_CPU_AVAILABLE

// WINDOW
#define JETSTREAM_MODULE_WINDOW_AVAILABLE
#define JETSTREAM_MODULE_WINDOW_CPU_AVAILABLE

// MULTIPLY
#define JETSTREAM_MODULE_MULTIPLY_AVAILABLE
#define JETSTREAM_MODULE_MULTIPLY_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_MULTIPLY_METAL_AVAILABLE */

// AMPLITUDE
#define JETSTREAM_MODULE_AMPLITUDE_AVAILABLE
#define JETSTREAM_MODULE_AMPLITUDE_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_AMPLITUDE_METAL_AVAILABLE */

// SCALE
#define JETSTREAM_MODULE_SCALE_AVAILABLE
#define JETSTREAM_MODULE_SCALE_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_SCALE_METAL_AVAILABLE */

// SOAPY
/* #undef JETSTREAM_MODULE_SOAPY_AVAILABLE */
/* #undef JETSTREAM_MODULE_SOAPY_CPU_AVAILABLE */

// AUDIO
#define JETSTREAM_MODULE_AUDIO_AVAILABLE
#define JETSTREAM_MODULE_AUDIO_CPU_AVAILABLE

// FM
#define JETSTREAM_MODULE_FM_AVAILABLE
#define JETSTREAM_MODULE_FM_CPU_AVAILABLE

// PAD
#define JETSTREAM_MODULE_PAD_AVAILABLE
#define JETSTREAM_MODULE_PAD_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_PAD_METAL_AVAILABLE */

// UNPAD
#define JETSTREAM_MODULE_UNPAD_AVAILABLE
#define JETSTREAM_MODULE_UNPAD_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_UNPAD_METAL_AVAILABLE */

// OVERLAP ADD
#define JETSTREAM_MODULE_OVERLAP_ADD_AVAILABLE
#define JETSTREAM_MODULE_OVERLAP_ADD_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_OVERLAP_ADD_METAL_AVAILABLE */

// INVERT
#define JETSTREAM_MODULE_INVERT_AVAILABLE
#define JETSTREAM_MODULE_INVERT_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_INVERT_METAL_AVAILABLE */

// MULTIPLY CONSTANT
#define JETSTREAM_MODULE_MULTIPLY_CONSTANT_AVAILABLE
#define JETSTREAM_MODULE_MULTIPLY_CONSTANT_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_MULTIPLY_CONSTANT_METAL_AVAILABLE */

// CAST
#define JETSTREAM_MODULE_CAST_AVAILABLE
#define JETSTREAM_MODULE_CAST_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_CAST_METAL_AVAILABLE */

// LINEPLOT
#define JETSTREAM_MODULE_LINEPLOT_AVAILABLE
#define JETSTREAM_MODULE_LINEPLOT_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_LINEPLOT_METAL_AVAILABLE */

// WATERFALL
#define JETSTREAM_MODULE_WATERFALL_AVAILABLE
#define JETSTREAM_MODULE_WATERFALL_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_WATERFALL_METAL_AVAILABLE */

// SPECTROGRAM
#define JETSTREAM_MODULE_SPECTROGRAM_AVAILABLE
#define JETSTREAM_MODULE_SPECTROGRAM_CPU_AVAILABLE
/* #undef JETSTREAM_MODULE_SPECTROGRAM_METAL_AVAILABLE */

// CONSTELLATION
#define JETSTREAM_MODULE_CONSTELLATION_AVAILABLE
#define JETSTREAM_MODULE_CONSTELLATION_CPU_AVAILABLE

// REMOTE
#define JETSTREAM_MODULE_REMOTE_AVAILABLE
#define JETSTREAM_MODULE_REMOTE_CPU_AVAILABLE

// AGC
#define JETSTREAM_MODULE_AGC_AVAILABLE
#define JETSTREAM_MODULE_AGC_CPU_AVAILABLE

// FOLD
#define JETSTREAM_MODULE_FOLD_AVAILABLE
#define JETSTREAM_MODULE_FOLD_CPU_AVAILABLE

// SPEECH RECOGNITION
/* #undef JETSTREAM_MODULE_SPEECH_RECOGNITION_AVAILABLE */
/* #undef JETSTREAM_MODULE_SPEECH_RECOGNITION_CPU_AVAILABLE */

// TAKE
#define JETSTREAM_MODULE_TAKE_AVAILABLE
#define JETSTREAM_MODULE_TAKE_CPU_AVAILABLE
```

luigifcruz, 2024-01-25:
Weird, the build system was supposed to stop you from compiling if the system couldn't be identified.

luigifcruz, 2024-01-25:
I added this patch to the `v1.0.0-alpha2` branch that should set the correct system flags for FreeBSD. Let me know if it works well for you!

yurivict, 2024-01-25:
> Weird, the build system was supposed to stop you from compiling if the system couldn't be identified.

I patched this place.

luigifcruz, 2024-01-27:
That makes more sense! 

Let me know if the patch works for you!

yurivict, 2024-01-28:
@luigifcruz 

meson.build should have ```target_machine.system().startswith('freebsd')```.

luigifcruz, 2024-02-20:
Thanks for checking this out. Implemented it in the `v1.0.0-alpha4` build script.

luigifcruz, 2024-02-24:
Please, re-open this issue if anything else seems broken in BSD.

closed 2024-02-24 by luigifcruz