luigifcruz/CyberEther #71: Catch2 minimum version

Zosoworld, 2024-02-20:
The latest version in the jammy repo of `Pop!_OS 22.04` of the catch2 library is `2.13.8-1`, and most of the headers are merged into `catch.hpp`:
[image]

The dependency in meson does not specify a minimum version, and thus when trying to build `libjetstream.so`, it does not find the include `<catch2/catch_session.hpp>`:
```bash
[110/122] Compiling C++ object tests/memory/jetstream-memory-types.p/types.cc.o
FAILED: tests/memory/jetstream-memory-types.p/types.cc.o 
ccache c++ -Itests/memory/jetstream-memory-types.p -Itests/memory -I../tests/memory -Iinclude -I../include -I. -I.. -I../include/jetstream/backend/devices/metal -I../include/jetstream/render/tools -Iresources -Iflowgraphs -Ishaders -I/usr/include/gstreamer-1.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/orc-0.4 -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -std=c++20 -O3 -pthread -MD -MQ tests/memory/jetstream-memory-types.p/types.cc.o -MF tests/memory/jetstream-memory-types.p/types.cc.o.d -o tests/memory/jetstream-memory-types.p/types.cc.o -c ../tests/memory/types.cc
```
Apparently, `catch_session.hpp` was renamed from `catch_session.h` around 4 years ago: [File History](https://github.com/catchorg/Catch2/commits/devel/src/catch2/catch_session.hpp), so a minimum version of around 3.0.0 should be built as a wrap to ensure it is build with LTS systems that might not have a newer version.

luigifcruz, 2024-02-20:
Thanks for the suggestion! I had to remove the WrapDB installation of Catch2 because it was installing its headers with CyberEther. But now I found a solution that prevent that from happening. It's currently live in the `v1.0.0-alpha4`. Let me know if that solves your problem!

Zosoworld, 2024-02-21:
That solved the issue with catch2. Thank you for the amazing work!

commit ec70efc0 (2024-02-20): Use wrap Catch2 in case Meson can't find system version >= 3.0. Fixes #71.
closed 2024-02-21 by Zosoworld