luigifcruz/CyberEther #72: Duplicate and Arithmetic modules using CUDA even though the backend is not present

Zosoworld, 2024-02-21:
Trying to compile the v1.0.0-alpha4 version, in Pop!_OS 22.04 with a Thinkpad L14 with AMD GPU, the CUDA Backend is of course not detected, but the Duplicate and Arithmetic modules try to use that backend, resulting in a compilation error:
[image]
[image]

Zosoworld, 2024-02-22:
Apologies for the issue, as the clone was very recent i just checked out the branch without pulling, without that last commit.
It compiled properly now, thank you for your time!

closed 2024-02-22 by Zosoworld

luigifcruz, 2024-02-22:
Grad it worked out fine!