luigifcruz/CyberEther #75: Build-browser errors (wgpu)

inwitio, 2024-03-14:
First time build & install for CyberEther (worked with no errors) and encountering errors on the browser-build. Installing on macOS.

CyberEther v1.0.0-alpha3-debugoptimized

```
[1/33] Compiling C++ object libjetstream.a.p/src_viewport_glfw_webgpu.cc.o
FAILED: libjetstream.a.p/src_viewport_glfw_webgpu.cc.o 
em++ -Ilibjetstream.a.p -I. -I.. -Iinclude -I../include -I../include/jetstream/backend/devices/metal -I../include/jetstream/render/tools -I../subprojects/SoapySDR-soapy-sdr-0.8.1/include -Isubprojects/SoapyAirspy-soapy-airspy-0.2.0 -I../subprojects/SoapyAirspy-soapy-airspy-0.2.0 -I../subprojects/airspyone_host-1.0.10/libairspy/src -I../subprojects/libusb-browser/include -Isubprojects/SoapyRTLSDR-soapy-rtl-sdr-0.3.3 -I../subprojects/SoapyRTLSDR-soapy-rtl-sdr-0.3.3 -I../subprojects/librtlsdr-1261fbb285297da08f4620b18871b6d6d9ec2a7b/include -Iresources -Iflowgraphs -Ishaders -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++20 -O2 -g -g -O3 -pthread -msimd128 -fexceptions -fPIC -DSOAPY_SDR_DLL_EXPORTS -MD -MQ libjetstream.a.p/src_viewport_glfw_webgpu.cc.o -MF libjetstream.a.p/src_viewport_glfw_webgpu.cc.o.d -o libjetstream.a.p/src_viewport_glfw_webgpu.cc.o -c ../src/viewport/glfw/webgpu.cc
**../src/viewport/glfw/webgpu.cc:104:15: error: no member named 'Release' in 'wgpu::SwapChain'**
  104 |     swapchain.Release();
      |     ~~~~~~~~~ ^
1 error generated.
```

```
[2/33] Compiling C++ object libjetstream.a.p/src_render_webgpu_window.cc.o
FAILED: libjetstream.a.p/src_render_webgpu_window.cc.o 
**../src/render/webgpu/window.cc:132:71: error: no member named 'Release' in 'wgpu::RenderPassEncoder'**
  132 |     auto renderPassEncoder = encoder.BeginRenderPass(&renderPassDesc).Release();
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
```

Please let me know what other information might be helpful for resolution.

luigifcruz, 2024-03-19:
There are two possible explanations here:

1. Your local Emscripten installation is older than the CyberEther implementation of WebGPU.

2. Your Emscripten installation is newer than the CyberEther implementation of WebGPU.

In any case, this is caused by the frequency of Emscripten changes in the WebGPU implementation. 

Can you tell us which Emscripten version are you trying to compile CyberEther with?

inwitio, 2024-04-08:
Thanks Luigi, I had the latest Emscripten installed 3.1.56. Looked around and saw 3.1.51 in the docker file and changed to that target. Compiled with no errors.

closed 2024-04-08 by inwitio

luigifcruz, 2024-04-08:
Glad it worked out! I plan to update this code and release it as alpha quality very soon. There are still some rough edges.