luigifcruz/CyberEther #76: glfwGrabErrorHandlerX11 bug preventing X11 GUI

drws, 2024-03-18:
Just built CyberEther on latest Arch Linux and there seems to be an issue with X11 GUI:

    > cyberether --no-hw-acceleration --no-validation --no-vsync
    JETSTREAM [INFO]  | -----------------------------------------------------
    JETSTREAM [INFO]  | Jetstream Heterogeneous Backend [VULKAN]
    JETSTREAM [INFO]  | -----------------------------------------------------
    JETSTREAM [INFO]  | Device Name:      AMD Radeon Graphics (RADV RENOIR)
    JETSTREAM [INFO]  | Device Type:      INTEGRATED
    JETSTREAM [INFO]  | API Version:      1.3.274
    JETSTREAM [INFO]  | Unified Memory:   NO
    JETSTREAM [INFO]  | Processor Count:  12
    JETSTREAM [INFO]  | Device Memory:    10.56 GB
    JETSTREAM [INFO]  | Staging Buffer:   64.00 MB
    JETSTREAM [INFO]  | Interoperability:
    JETSTREAM [INFO]  |   - Can Import Device Memory: YES
    JETSTREAM [INFO]  |   - Can Export Device Memory: YES
    JETSTREAM [INFO]  |   - Can Export Host Memory:   YES
    JETSTREAM [INFO]  | -----------------------------------------------------
    cyberether: /usr/src/debug/glfw/glfw-3.4/src/x11_init.c:1099: _glfwGrabErrorHandlerX11: Assertion `_glfw.x11.errorHandler == NULL' failed.
    Aborted (core dumped)

`cyberether --headless` works, so the issue seems to be in the X11 part of the code.

luigifcruz, 2024-03-19:
Looks like there is a problem with GLFW. This is weird because I never saw this error before. Are you sure the GLFW headers and libraries match? 

In any case, building CyberEther with the internal GLFW should fix this problem. To do this, just make sure Meson can't find the installed GLFW version.

drws, 2024-03-20:
There's only a single glfw package on Arch, so I believe headers and libraries should match. And no open issues regarding the package. Also tried to rebuid the package to no avail, but didn't try the internal glfw yet. Do you have any easy way in mind to prevent meson from finding glfw?

luigifcruz, 2026-04-19:
This is likely solved in the newest version. Please re-open if not.