luigifcruz/CyberEther #80: Multichannel source support

vd2org, 2024-07-03:
Hi! We have hardware with both 2 and 8 channels. 
Currently, the SOA0 block supports only a single channel.

Have you plans to support more channels ?

luigifcruz, 2024-07-03:
Hey! 

Are the same rate of every channel the same? 

I have a solution to this use case in mind. But it would only work if the sample rate of every channel is the same.

vd2org, 2024-07-04:
Yes, at this time all boards working on the same sample rate and the same frequency.
Some other parameters(like gain) might be different but, this is not necessary at this time.

luigifcruz, 2024-07-05:
Nice! I think my idea should work then.

In short, the idea is to modify the Soapy module to create an array with an extra dimension for the SDR channel. For example, the single channel module creates a tensor with shape [8, 8192]. With the first dimension being the batch size and the second the number of elements. When operating with multiple streams, the generated tensor would be [2, 8, 8192]. The tensor could be broken down externally with a splice block to be used however you want. 

I should implement this feature at some point depending on my free time. But you are welcome to take a crack at it. I don't expect it to be very difficult to implement.