luigifcruz/CyberEther #81: 22.04 Ubuntu aarch64 Vulkan Error VK_ERROR_INITIALIZATION_FAILED

alphafox02, 2024-08-05:
On a 22.04 aarch64 build on a Pi4 I've ran into the following using the main and alpha 5 branches (running it with or without root).

ubuntu@ubuntu:/usr/src/CyberEther/build$ ./cyberether
JETSTREAM [WARN]  | [VULKAN] Optional device extension 'VK_EXT_external_memory_host' is not supported.
JETSTREAM [WARN]  | [VULKAN] Optional device extension 'VK_KHR_portability_subset' is not supported.
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [INFO]  | Jetstream Heterogeneous Backend [VULKAN]
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [INFO]  | Device Name:      V3D 4.2.14
JETSTREAM [INFO]  | Device Type:      INTEGRATED
JETSTREAM [INFO]  | API Version:      1.2.255
JETSTREAM [INFO]  | Unified Memory:   YES
JETSTREAM [INFO]  | Processor Count:  4
JETSTREAM [INFO]  | Device Memory:    1.85 GB
JETSTREAM [INFO]  | Staging Buffer:   64.00 MB
JETSTREAM [INFO]  | Interoperability:
JETSTREAM [INFO]  |   - Can Import Device Memory: YES
JETSTREAM [INFO]  |   - Can Export Device Memory: YES
JETSTREAM [INFO]  |   - Can Export Host Memory:   NO
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [ERROR] | [VULKAN] Can't create swapchain.
JETSTREAM [ERROR] | [VULKAN] Error code: VK_ERROR_INITIALIZATION_FAILED
Function main (../main.cc@180) throwed!
terminate called after throwing an instance of 'Jetstream::Result'
Aborted (core dumped)

luigifcruz, 2024-08-05:
Weird, last time I tried to run this on a Pi it worked well. Maybe I screwed up some config. 

Can you set the following envoriment variable and post the logs again, please?

```
export JST_DEBUG=3
```

alphafox02, 2024-08-06:
I rebuilt directly on the pi so I could get more info (took forever)

Here's more info which I'll be looking at what I can change, if anything.

```
ubuntu@ubuntu:~/CyberEther/build$ ./cyberether
JETSTREAM [DEBUG] | [SCHEDULER] Initializing compute graph.
JETSTREAM [DEBUG] | [INSTANCE] Creating instance.
JETSTREAM [DEBUG] | [INSTANCE] Building interface
JETSTREAM [DEBUG] | Initializing Vulkan backend.
JETSTREAM [DEBUG] | [VULKAN] Required instance extensions: {"VK_EXT_debug_report", "VK_KHR_surface", "VK_KHR_xcb_surface"}
JETSTREAM [DEBUG] | [VULKAN] Optional instance extensions: {}
JETSTREAM [DEBUG] | [VULKAN] Candidate device - Extension check: OK, Queue family check: OK
JETSTREAM [DEBUG] | [VULKAN] Candidate device - Extension check: OK, Queue family check: OK
JETSTREAM [DEBUG] | [VULKAN] Required device extensions: {"VK_KHR_swapchain"}
JETSTREAM [DEBUG] | [VULKAN] Optional device extensions: {"VK_EXT_external_memory_host", "VK_KHR_external_memory_fd", "VK_KHR_portability_subset"}
JETSTREAM [WARN]  | [VULKAN] Optional device extension 'VK_EXT_external_memory_host' is not supported.
JETSTREAM [WARN]  | [VULKAN] Optional device extension 'VK_KHR_portability_subset' is not supported.
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [INFO]  | Jetstream Heterogeneous Backend [VULKAN]
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [INFO]  | Device Name:      V3D 4.2.14
JETSTREAM [INFO]  | Device Type:      INTEGRATED
JETSTREAM [INFO]  | API Version:      1.2.255
JETSTREAM [INFO]  | Unified Memory:   YES
JETSTREAM [INFO]  | Processor Count:  4
JETSTREAM [INFO]  | Device Memory:    1.85 GB
JETSTREAM [INFO]  | Staging Buffer:   64.00 MB
JETSTREAM [INFO]  | Interoperability:
JETSTREAM [INFO]  |   - Can Import Device Memory: YES
JETSTREAM [INFO]  |   - Can Export Device Memory: YES
JETSTREAM [INFO]  |   - Can Export Host Memory:   NO
JETSTREAM [INFO]  | -----------------------------------------------------
JETSTREAM [DEBUG] | [VULKAN] Creating GLFW viewport.
vulkan: No DRI3 support detected - required for presentation
Note: you can probably enable DRI3 in your Xorg config
JETSTREAM [DEBUG] | [VULKAN] Validation Error: [ VUID-vkGetPhysicalDeviceSurfaceCapabilitiesKHR-surface-06211 ] Object 0: handle = 0xaaab118ee760, type = VK_OBJECT_TYPE_PHYSICAL_DEVICE; | MessageID = 0x11fe03f4 | vkGetPhysicalDeviceSurfaceCapabilitiesKHR(): surface is not supported by the physicalDevice. The Vulkan spec states: surface must be supported by physicalDevice, as reported by vkGetPhysicalDeviceSurfaceSupportKHR or an equivalent platform-specific mechanism (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-vkGetPhysicalDeviceSurfaceCapabilitiesKHR-surface-06211)                  
JETSTREAM [DEBUG] | [VULKAN] Validation Error: [ VUID-vkGetPhysicalDeviceSurfaceFormatsKHR-surface-06211 ] Object 0: handle = 0xaaab118ee760, type = VK_OBJECT_TYPE_PHYSICAL_DEVICE; | MessageID = 0x706bf88b | vkGetPhysicalDeviceSurfaceFormatsKHR(): surface is not supported by the physicalDevice. The Vulkan spec states: surface must be supported by physicalDevice, as reported by vkGetPhysicalDeviceSurfaceSupportKHR or an equivalent platform-specific mechanism (https://www.khronos.org/registry/vulkan/specs/1.3-khr-extensions/html/vkspec.html#VUID-vkGetPhysicalDeviceSurfaceFormatsKHR-surface-06211)                             
JETSTREAM [DEBUG] | [VULKAN] Validation Error: [ VUID-vkGetPhysicalDeviceSurfacePresentModesKHR-surface-06211 ] Object 0: handle = 0xaaab118ee760, type = VK_OBJECT_TYPE_PHYSICAL_DEVICE; | MessageID = 0x504cc285 | vkGetPhysicalDeviceSurfacePresentModesKHR(): surface is not supported by the physicalDevice. The Vulkan spec states: surface must be supported by physicalDevice, as reported by vkGetPhysicalDeviceSurfaceSupportKHR or an equivalent platform-specific mechanism (https://www.khronos.org/registry/vulkan/specs/1.3-khr-extensions/html/vkspec.html#VUID-vkGetPhysicalDeviceSurfacePresentModesKHR-surface-06211)              
JETSTREAM [DEBUG] | [VULKAN] Validation Error: [ VUID-vkGetPhysicalDeviceSurfaceFormatsKHR-surface-06211 ] Object 0: handle = 0xaaab118ee760, type = VK_OBJECT_TYPE_PHYSICAL_DEVICE; | MessageID = 0x706bf88b | vkGetPhysicalDeviceSurfaceFormatsKHR(): surface is not supported by the physicalDevice. The Vulkan spec states: surface must be supported by physicalDevice, as reported by vkGetPhysicalDeviceSurfaceSupportKHR or an equivalent platform-specific mechanism (https://www.khronos.org/registry/vulkan/specs/1.3-khr-extensions/html/vkspec.html#VUID-vkGetPhysicalDeviceSurfaceFormatsKHR-surface-06211)                             
JETSTREAM [DEBUG] | [VULKAN] Validation Error: [ VUID-vkGetPhysicalDeviceSurfacePresentModesKHR-surface-06211 ] Object 0: handle = 0xaaab118ee760, type = VK_OBJECT_TYPE_PHYSICAL_DEVICE; | MessageID = 0x504cc285 | vkGetPhysicalDeviceSurfacePresentModesKHR(): surface is not supported by the physicalDevice. The Vulkan spec states: surface must be supported by physicalDevice, as reported by vkGetPhysicalDeviceSurfaceSupportKHR or an equivalent platform-specific mechanism (https://www.khronos.org/registry/vulkan/specs/1.3-khr-extensions/html/vkspec.html#VUID-vkGetPhysicalDeviceSurfacePresentModesKHR-surface-06211)              
JETSTREAM [DEBUG] | [VULKAN] Swap mailbox presentation mode is available.
[... cut ...] e a surface that is supported by the device as determined using vkGetPhysicalDeviceSurfaceSupportKHR (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkSwapchainCreateInfoKHR-surface-01270)                 
JETSTREAM [ERROR] | [VULKAN] Can't create swapchain.
JETSTREAM [ERROR] | [VULKAN] Error code: VK_ERROR_INITIALIZATION_FAILED
Function main (../main.cc@180) throwed!
terminate called after throwing an instance of 'Jetstream::Result'
Aborted (core dumped)

```

luigifcruz, 2024-08-10:
Looks like it's a Vulkan driver issue. Most likely an installation problem. 

Can you try to run `vkcube`? If it works we need to do further testing.